Auterion/px4-jsbsim-bridge #21: Rotor torque being multiplied by a moment arm?

seanmcleod, 2020-10-13:
In the hexarotor and quadrotor models why is the net torque being multiplied by a moment arm?

https://github.com/Auterion/px4-jsbsim-bridge/blob/263da94dfd74355e6ddc9298caab9fec306d3ec7/models/hexarotor_x/hexarotor_x.xml#L586-L597

Jaeyoung-Lim, 2020-10-14:
@mvacanti Could you share any insights?

mvacanti, 2020-10-14:
@seanmcleod & @Jaeyoung-Lim Good catch. This is a holdover from previous work.  Should be:

```xml
 <axis name="YAW"> 
     <function name="aero/rotor_yaw"> 
     <description>Rotor Yaw</description> 
     <sum> 
         <property>fcs/net_torque</property> 
         <property>fcs/torque_bypass</property> 
     </sum> 
   </function> 
 </axis> 
```

seanmcleod, 2020-10-14:
What are `fcs/torque_bypass` and `fcs/lift_bypass[x]` for?

mvacanti, 2020-10-14:
@seanmcleod these were properties I was using to test performance in JSBSim via telnet / scripts to evaluate model performance. It allowed me to 1. Isolate PX4 outputs in respect to vehicle movement (hold vehicle stationary or cause it to move in manner controlled explicitly by JSBSim) 2. Introduce disruptions to observe PX4 response. Can be simplified but believe the above are valuable uses.

seanmcleod, 2020-10-14:
@mvacanti so maybe they're not named ideally? Maybe something like `external_torque_disturbance`?

I had searched through the rest of the model files and also the bridge source code for references and couldn't find any.

seanmcleod, 2020-10-14:
@mvacanti yep I meant since I couldn't find any references to them in the rest of the repo it was difficult to figure out what they were being used for so was just guessing based on the `bypass` in the names.

Wouldn't necessarily remove them, rather add a comment in the xml explaining what they can be used for and possibly rename them.