Auterion/px4-jsbsim-bridge #27: ENABLING HIL MODE IN JSBSIM BRIDGE

xfahad100, 2020-10-22:
it seems like it is only sitl enabled. I have successfully enabled serial by changing mavlink tcp port to ttyACM0 in the configs file.
I want to enable HIL mode and serial communication at the same time. a little help would be appreciated. thanks

Jaeyoung-Lim, 2020-10-22:
@xfahad100 I created a draft pull request from the WIP branch I was working on in #28 

Please feel free to move it forward

xfahad100, 2020-10-23:
serial connectivity is resolved, but for hitl simulation, connectivity of gcs udp port is also required.

xfahad100, 2020-10-23:
hil mode is achieved. px4 is sending data through mavlink to qgc in hil simulation

xfahad100, 2020-10-23:
I will close this issue in 24 hours, thanks for your time.

Jaeyoung-Lim, 2020-10-23:
@xfahad100 Awesome, so the PR just worked out of the box for you?

xfahad100, 2020-10-23:
@Jaeyoung-Lim No , I had to make some changes related to connectivity with Qgroundcontrol.

xfahad100, 2020-10-23:
They were very minor changes . all credit goes to you! @Jaeyoung-Lim

Jaeyoung-Lim, 2020-10-23:
@xfahad100 Would you mind making a PR with your changes included?

xfahad100, 2020-10-24:
@Jaeyoung-Lim Yes I will in few days! I just need to comment outunnecessary changes.

Jaeyoung-Lim, 2020-10-24:
@xfahad100 Awesome, thanks! I will keep this open until it is fully resolved and merged :smile:

xfahad100, 2020-10-25:
#35 #36 merge them @Jaeyoung-Lim

Jaeyoung-Lim, 2020-10-25:
@xfahad100 Is there a reason those two are separated in two PRs?

xfahad100, 2020-10-25:
@Jaeyoung-Lim uhh no,, I'm a newbie on github and I cant figure out  how to compare changes that I've made in 2 files. So there are 2 prs for each file ..... sorry for the mess

Jaeyoung-Lim, 2020-10-25:
@xfahad100 I tried your PR, but was not able to get GPS position. Were you able to fly with the changes you made?

xfahad100, 2020-10-25:
ACM0 is workin fine but USB0(telemetry) is giving tx overflow error.

Jaeyoung-Lim, 2020-10-25:
> ACM0 is workin fine

Can you confirm that you can fly with the vehicle in the simulation? which airframe were you using?

Jaeyoung-Lim, 2020-12-22:
@xfahad100 Any updates?

xfahad100, 2021-01-27:
It starts flying by itself in HIL mode and does not respond to px4 data....

Jaeyoung-Lim, 2021-01-27:
@xfahad100 Are you saying that it didn't work for you?

xfahad100, 2021-02-10:
@Jaeyoung-Lim  is mag sensor plugin not working at the moment?
I will share my results once I am able to fly my vehicle.

Jaeyoung-Lim, 2021-02-10:
@xfahad100 Now I am a bit confused. You are saying it worked before and not anymore?

xfahad100, 2021-02-10:
@Jaeyoung-Lim bruhh I never said that.. I am now able to get gps location but I am not able to fly ..

Jaeyoung-Lim, 2021-02-10:
@xfahad100 Yes exactly. I have asked you multiple times if it was working, but you never answered. So are you saying it has never worked?

xfahad100, 2021-02-10:
I'm working on it n its almost done...

Jaeyoung-Lim, 2021-02-11:
@xfahad100 I am assuming that means it never worked for you.

xfahad100, 2021-02-17:
@Jaeyoung-Lim it is working for me right now

xfahad100, 2021-02-17:
At least it takes input from rc and throw it out to jsbsim

xfahad100, 2021-11-03:
Bro I am doing hitl successfully with px4 and jsbsim and also implementing autopilot

xfahad100, 2021-11-03:
But I have excluded qgroundcontrol