Auterion/px4-jsbsim-bridge #42: Error in make jsbsim in linux

moreba1, 2020-12-30:
bayanlou@ubuntu:~/PX4-Autopilot$ make px4_sitl jsbsim

[0/4] Performing build step for 'jsbsim_bridge'
[1/2] Building CXX object CMakeFiles/j...sim_bridge.dir/src/jsbsim_bridge.cpp.o
FAILED: CMakeFiles/jsbsim_bridge.dir/src/jsbsim_bridge.cpp.o 
/usr/bin/c++  -DJSBSIM_ROOT_DIR=\"/home/bayanlou/PX4-Autopilot/Tools/jsbsim_bridge\" -I/home/bayanlou/PX4-Autopilot/mavlink/include -I/usr/include/JSBSim -I/usr/include/eigen3 -I/home/bayanlou/PX4-Autopilot/Tools/jsbsim_bridge/include -std=c++14 -pthread   -Wno-address-of-packed-member --warn-uninitialized -MD -MT CMakeFiles/jsbsim_bridge.dir/src/jsbsim_bridge.cpp.o -MF CMakeFiles/jsbsim_bridge.dir/src/jsbsim_bridge.cpp.o.d -o CMakeFiles/jsbsim_bridge.dir/src/jsbsim_bridge.cpp.o -c /home/bayanlou/PX4-Autopilot/Tools/jsbsim_bridge/src/jsbsim_bridge.cpp
/home/bayanlou/PX4-Autopilot/Tools/jsbsim_bridge/src/jsbsim_bridge.cpp: In member function ‘bool JSBSimBridge::SetFdmConfigs(ConfigurationParser&)’:
/home/bayanlou/PX4-Autopilot/Tools/jsbsim_bridge/src/jsbsim_bridge.cpp:146:69: error: cannot convert ‘std::shared_ptr<JSBSim::FGInitialCondition>’ to ‘JSBSim::FGInitialCondition*’ in initialization
     JSBSim::FGInitialCondition *initial_condition = _fdmexec->GetIC();
                                                                     ^
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-address-of-packed-member’
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/jsbsim_bridge/jsbsim_bridge-build 
cd /home/bayanlou/PX4-Autopilot/build/px4_sitl_default/build_jsbsim_bridge && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

Jaeyoung-Lim, 2020-12-30:
A friendly reminder for @moreba1 to provide detailed instructions on what environment, system you are running in and how to reproduce the problem.

This has been a repeated pattern on different repos from @moreba1 posting incomplete issues and then asking for more detailed information in order to figure out what the problem is. It would save both @moreba1 and my time if a detailed instruction would be written with the issue from the start of the conversation. 

The problem is most likely coming from your local system since this repo is verifying the builds in a automated CI: https://github.com/Auterion/px4-jsbsim-bridge/actions?query=workflow%3A%22Firmware+Build+Tests%22

moreba1, 2021-01-04:
this release version worked JSBSim-devel_1.1.2-172.bionic.amd64.deb
https://github.com/JSBSim-Team/jsbsim/releases

closed 2021-01-04 by moreba1