Auterion/px4-jsbsim-bridge #46: RLException when calling roslaunch on jsbsim_bridge

zimmy87, 2021-07-29:
I followed the instructions for running the bridge with ROS [here](https://github.com/Auterion/px4-jsbsim-bridge#running-the-bridge-with-ros). The bridge package appears to build correctly but when I go to launch it with:

`roslaunch jsbsim_bridge px4_jsbsim_bridge.launch`

I get the following output:

```
RLException: [px4_jsbsim_bridge.launch] is neither a launch file in package [jsbsim_bridge] nor is [jsbsim_bridge] a launch file name
The traceback for the exception was written to the log file
```

Does anybody know what could be going wrong here?

Jaeyoung-Lim, 2021-07-30:
@zimmy87 Have you sourced yout workspace?

zimmy87, 2021-08-09:
Thanks @Jaeyoung-Lim! That solved the RLException error I'm seeing, but now when I run the `roslaunch jsbsim_bridge px4_jsbsim_bridge.launch` command I get the following output:

```
... logging to /home/v-dazi/.ros/log/c06e3c44-f8b5-11eb-950c-00155d59a2df/roslaunch-LAPTOP-726PT48T-1778.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: px4
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/home/v-dazi/catkin_ws/src/px4-jsbsim-bridge
ROS path [2]=/home/v-dazi/AirSim/ros/src
ROS path [3]=/opt/ros/noetic/share
The traceback for the exception was written to the log file
```

The only web search result I could find mentioning the `Resource not found: px4` error is [here](https://github.com/PX4/PX4-Avoidance/issues/596), but all the solutions mentioned in that thread were for a Gazebo-specific scenario and didn't work for me locally. Do you know what could be going on here?

Jaeyoung-Lim, 2021-08-09:
@zimmy87 Great, this means that you don't have the PX4 catkin package in your path.

Where is your PX4 located? Probably the documentation is assuming that you have a working PX4 SITL setup.

For example, you need to do (http://docs.px4.io/master/en/simulation/ros_interface.html)
```
cd <PX4-Autopilot_clone>
DONT_RUN=1 make px4_sitl_default gazebo
source ~/catkin_ws/devel/setup.bash    # (optional)
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 posix_sitl.launch
```

Jaeyoung-Lim, 2021-10-12:
@zimmy87 Please share the whole log.