Auterion/px4-jsbsim-bridge #48: "Unable to start mission: Vehicle rejected arming." when taking off

zimmy87, 2021-09-20:
When running the PX4 JSBSim bridge via `make px4_sitl jsbsim` on Ubuntu 18.04 and with QGroundControl connected, I get a error dialog stating `Unable to start mission: Vehicle rejected arming.` after uploading a mission and trying to take off. This doesn't seem to be fatal, as the mission can still proceed afterwards, but considering I don't see any options for arming/disarming inside QGroundControl, is there anything I can do about this error?

Jaeyoung-Lim, 2021-09-27:
@zimmy87 You seem to be having gps related issues as written in the console.

Also the fact that you don't see any terrain in flightgear seems wrong to me. Can you see the airport terrain when you start flightgear?

Jaeyoung-Lim, 2021-11-03:
@zimmy87 Any updates regarding this issue?