Auterion/px4-jsbsim-bridge #49: "Airspeed sensor failure detected (1, -1)" after landing

zimmy87, 2021-09-20:
When running the PX4 JSBSim bridge via `make px4_sitl jsbsim` on Ubuntu 18.04 and with QGroundControl connected, I'm seeing the PX4 console output  a `poll timeout 0, 22` error after completing a mission and landing, as well as a `Airspeed sensor failure detected (1, -1)` dialog inside QGroundControl. The view inside of FlightGear goes black and PX4 is unresponsive at this point and needs a restart before uploading another mission. 
[image]

Jaeyoung-Lim, 2021-09-20:
@zimmy87 Thanks for reporting, I will have a look

Jaeyoung-Lim, 2021-11-03:
@zimmy87 Turns out this is not something coming from the simulator, but was a real issue on the flight controller side.

This was reported in https://github.com/PX4/PX4-Autopilot/issues/18437 and fixed by https://github.com/PX4/PX4-Autopilot/pull/18442

Can you confirm that the issue has also been fixed?

rrajnidhi, 2022-04-13:
Even i was facing same issue. I guess the reason was bcz of steady airspeed sensor value getting published. PX4 considers this as a sensor fault. I have resolved it by changing the value of wait time from 1 second to 10second in /src/modules/airspeed_selector/airspeed_selector_main.cpp
Try and let me know.

Jaeyoung-Lim, 2022-04-13:
@rrajnidhi Are you sure your firmware includes the fix in https://github.com/PX4/PX4-Autopilot/pull/18442 ?