Auterion/px4-jsbsim-bridge #51: ERROR [commander] Critical failure detected: lockdown

Jaeyoung-Lim, 2021-10-10:
**Problem Description**
Trying to takeoff using the default rascal model results in a critical lockdown

**To Reproduce**
```
make px4_sitl jsbsim
```
Error Log: 
```
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2021-10-10/19_52_41.ulg	
INFO  [logger] Opened full log file: ./log/2021-10-10/19_52_41.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [commander] Failsafe mode deactivated	
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative

pxh> 
pxh> commander takeoff
pxh> INFO  [commander] Armed by internal command	
INFO  [navigator] Using minimum takeoff altitude: 20.00 m	
INFO  [fw_pos_control_l1] Takeoff on runway	
INFO  [commander] Takeoff detected	
ERROR [commander] Critical failure detected: lockdown	
INFO  [runway_takeoff] #Takeoff airspeed reached	
INFO  [commander] Disarmed by lockdown	
INFO  [logger] closed logfile, bytes written: 3148319
WARN  [ekf2] primary EKF changed 0 (gyro fault) -> 4

```

zimmy87, 2021-10-19:
When we tested with JSBSim v1.1.0, we didn't see this issue.

tayyabkhalil-313, 2022-11-16:
@Jaeyoung-Lim Were you able to solve this? I am getting the same issue with JSBSim 1.2.0 and PX4 1.12.3.