Auterion/px4-jsbsim-bridge #53: JSBSIM not found

pbecchi, 2021-12-18:
Hi,
I succesfully compile and build JSBSim in Ubuntu 32 bit.

When I try to build PX4:
`HEADLESS=1 make px4_sitl jsbsim_quadrotor_x`
i got the message JSBSIM not found.

I have updated  $PATH adding JSBSim directory ("/jsbsim/build/src") but this doesnt solve the problem.
Thanks for your help.

Jaeyoung-Lim, 2021-12-18:
@pbecchi Can you confirm that you have installed jsbsim?

pbecchi, 2021-12-18:
Yes, It has been built and tested using the line command. 
Initially I thought that "not found" was due to $PATH , but now the jsbsim directory has been added to it .....
The message is still there. 
Thanks

pbecchi, 2021-12-23:
I have found that jsbsim was not correctly installed.
Now px4-sitl build succesfully!
Thanks

closed 2021-12-23 by pbecchi

Jaeyoung-Lim, 2021-12-23:
@pbecchi Thanks for the update!