Auterion/px4-jsbsim-bridge #54: multi vehicle support in px4 jsbsim-bridge

airsimdevdd, 2021-12-22:


Jaeyoung-Lim, 2021-12-27:
@airsimdevdd Could you please clarify what multivehicle support means in this context?

Are you trying to run multiple vehicles in a single jsbsim instance? or are you just talking about simulating multiple vehicles at once in parallel?

if it is the latter, you can already do this by creating a script to launch multiple instances of SITL

airsimdevdd, 2021-12-28:
simulating multiple vehicles at once in parallel sir

Jaeyoung-Lim, 2021-12-28:
@airsimdevdd could you clarify what does at once mean? The two examples are both simulating "at once"

airsimdevdd, 2021-12-28:
Sir ,I want to fly two or more vehicle simultaneously in sitl

Jaeyoung-Lim, 2022-10-18:
@ParamDeshpande You probably need multiple instances of jsbsim running in order to simulate multiple vehicles

ParamDeshpande, 2022-11-03:
Thanks ! Got the SITL multi-plane simulation up and running.

closed 2023-03-15 by airsimdevdd