Auterion/px4-jsbsim-bridge #55: Custom quadrotor model

pbecchi, 2021-12-23:
I am trying to add dj f450 model to px4 jsbsim bridge.

I have a jsbsim model that performed succesfully some test https://github.com/JSBSim-Team/jsbsim/issues/333 and i would like now to bridge it to px4 sitl.
Dj f450 already exist as a PX4 copter example, so i guess that i need only to move all xml files to  tools/jsbsim-bridge/models
For sure i need to interface correctely motor inputs  to fcs commands coming from PX4 sitl .  What esle do i need to do?
.Where i can find guidance or examples?

Jaeyoung-Lim, 2021-12-27:
@pbecchi You can already look at the quadrotor model as an example in this repository.

pbecchi, 2021-12-27:
Yes, thanks, it is what I am doing.
I tell where I am:
I have made following file modifications or additions ("airframe" stand for the name of the new model.
<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 15">
</head>

<body lang=en-GB style='font-family:Calibri;font-size:11.0pt'>

<div style='direction:ltr'>

Add   "airframe" jsbsim model files in : tools/jsbsim_bridge/models | Added   dj_f450 directory added engine directory
-- | --
Add   "airframe".xml file to tools/jsbsim_bridge/configs | Added   dj_f450.xsl identical to quadrotor.xml
Add   "XXXX_airframe" to   ROMFS/px4fmu_ommon/init.d-posix/airframes | Added   4011_dj_f450
Add to   CMakeList.txt line with "XXXX_airframe" | Added   4011_dj_f450
Modify   setup_jsbsim.bash in tools/                adding case "aircraft)" with    MODEL_NAME="aircraft" | Added    dj_f450)               MODEL_NAME="dj_f450"               ;;
Modify   sitl_target_cmake in platforms/posix/cmake | Under  set(models_jsbsim     Added line          dj_f450

</div>

</body>

</html>

Than I succesfully build  PX4 with : HEADLESS=1 make PX4_sitl  jsbsim_djf450,
When execution start PX4 start with no error . JSBSIM start and read model.xml files.
PX4 wait for simulator feedbach with no answer from JSBSIM. Last maessage is :
<INFO [simulator] Waiting for simulator to accept connection on TCP port 4560>

I appreciate any suggestion or help to keep going .... Thanks

Jaeyoung-Lim, 2021-12-27:
> When execution start PX4 start with no error . JSBSIM start and read model.xml files.

Are you sure that your jsbsim model has been started? Something is not starting properly on the jsbsim simulation side and unable communicate with PX4

pbecchi, 2021-12-27:
Correct! it look like jsbsim is not able to find the engine file.
Engines and propeller files are normally located on an on a separate /engine/ directory that I have created in tools/jsbsim_bridge/.
It is the right location or the engines files should be in the aircraft directory?

pbecchi, 2022-01-04:
I have solved the problen of the link to engine file:
the file containing engines definitions that is /engine/ on jsbsim , here is /Engines/. It took some time to find the difference...
Now I am able to run simulations!!
My setup is an old 32bit PC with Ubuntu where I run PX4_sitl and JSBSim. It is connected via local lan to a windows laptop were QGC run that is connected via USB to a FRSKY radio trasmitter used as a joystick.
Quadrotor_x  flight OK  ; results seams as expected.
My new model dj_f450 (the well known DJI model) use a JSBSim model I Have tested with lauch scripts . it seams to work also on PX4_sitl. I have seen from ULG logs that pitch and roll attitude seams to change consistently to joystick commands but position seams not to change.
There is any way to LOG JSBSim variable values? This is what i did testing the model with the JSBSim scritp:
`<output file="OutputFgfs.xml"></output>
  <output name="quad_log.csv" type="CSV" rate="250">
        <property>position/h-agl-ft</property>
        <property>velocities/vc-kts</property>
        <property>attitude/phi-rad</property>
        <property>attitude/theta-rad</property>
        <property>attitude/psi-rad</property>
        <property>propulsion/engine[0]/propeller-rpm</property>
  	<property>propulsion/engine[0]/thrust-lbs</property>
  	<property>propulsion/engine[0]/power-hp</property>
        <property>propulsion/engine[1]/propeller-rpm</property>
  	<property>propulsion/engine[1]/thrust-lbs</property>
  	<property>propulsion/engine[1]/power-hp</property>
`

pbecchi, 2022-01-17:
I am working on preparation of a PR with a dji f450 quadrotor example.
I am trying to understand the meaning of `<actuators>` definitions on `/configs/` files.
I am confused because in jsbsim there is a class `<actuator>` that i dont think has nothing to do with it.
I guess it is a description of the FCS interface between px4_sitl and jsbsim , but i dont know the meaning of various elements e.g.:
in quadrotor.xml:
` <channel name="rotor0">
            <index>0</index>
            <scale>1</scale>
            <property>fcs/esc-cmd-norm[0]</property>
        </channel>`
in rascal.xml:
`<channel name="aileron">
            <index>5</index>
            <scale>-1</scale>
            <property>fcs/aileron-cmd-norm</property>
        </channel>`
Where I can find any documentation? 
Or what is the meaning of <index> and <scale> and where i can find the properties lists?

pbecchi, 2022-01-17:
Let me guess what could be the answer:
each `<channel>` assign PX4 `pwm_output0[ index]` to the property identified. The output will be multiplied by the scale factor.
Remaining questions :
It is possible to assign only pwm_output0 so only 8 properties?
Or pwm_output1 it is also adressable?