Auterion/px4-jsbsim-bridge #57: Failure to build jsbsim_bridge using catkin

oricabel, 2022-06-12:
Hello all!
I am currently trying to run the jsbsim simulation using the PX4 jsbsim bridge but during the bulid, one package fails.

input: "
$ catkin build jsbsim_bridge
"

output: "
Failed << jsbsim_bridge:make [ Exited with code 2 ]
Failed <<< jsbsim_bridge [ 1.4 seconds ]
[build] Summary: 4 of 5 packages succeeded.
[build] Ignored: 2 packages were skipped or are skiplisted.
[build] Warnings: None.
[build] Abandoned: None.
[build] Failed: 1 packages failed.
[build] Runtime: 3.9 seconds total. "

Please, can anyone shed some light as to why the build fails?
I am using ubuntu 18.04, I have followed "ROS with MAVROS installation guide" and used ROS melodic.

Here's the log (I hope it is the correct one)
[build.make.log](https://github.com/Auterion/px4-jsbsim-bridge/files/8886302/build.make.log)

Thanks!

Jaeyoung-Lim, 2022-06-13:
@oricabel Are you trying to use this package as an individual catkin package?

The build log doesnt seem to show why it failed. Are you sure that is the full log?

oricabel, 2022-06-13:
I ran it using "catkin build -v" and received a similar error but slightly different outpu
[build.make.log](https://github.com/Auterion/px4-jsbsim-bridge/files/8893662/build.make.log)

And yes, the steps written in the readme

Jaeyoung-Lim, 2022-06-13:
@oricabel If you look at the log carefully, you can see the compilation error:
```
common/mavlink.h: No such file or directory
```
The mavlink is found through cmake:
https://github.com/Auterion/px4-jsbsim-bridge/blob/f88764b091a8f5bc5f98893f07c63b56682a1420/CMakeLists.txt#L7

You either need to install mavlink, or link it properly. Have you already built mavlink or PX4?

oricabel, 2022-06-14:
I don't think I built mavlink, and I am not sure about jsbsim but I did install a jsbsim release. (I am not entirely sure what it means to "build").
So, do I use the CmakeLists file to install mavlink? and if so, can you tell me how to do that?

Jaeyoung-Lim, 2022-06-15:
> I am using ubuntu 18.04, I have followed "ROS with MAVROS installation guide" and used ROS melodic.

@oricabel I would remove jsbsim_bridge from your workspace, and make sure that everything else works first. If you have mavros built, mavlink should be already in your workspace.

Also can you run jsbsim without building it through ROS? (running with `make px4_sitl jsbsim` as written in https://docs.px4.io/v1.12/en/simulation/jsbsim.html)

oricabel, 2022-06-22:
I can run jsbsim without building it through ROS, I the thought the ROS could help me load the aerodynamic model i created, but I'm not so sure now... Thank you so much for the help!

shupx, 2024-09-08:
```bash
sudo apt install ros-noetic-mavlink
```
Then change `#include <common/mavlink.h>` into `#include <mavlink/v2.0/common/mavlink.h>`. Run `catkin_make` in the workspace.