Auterion/px4-jsbsim-bridge #58: Loading a custom aircraft model to jsbsim bridge

oricabel, 2022-06-22:
Hello,
I have an aerodynamic model that I wrote in .XML format and I want to simulate it using the bridge, but I don't understand how.
I tried to copy the model's folder into a folder with the rest of the models that work (Rascal, Malolo, etc..), but when I try to run it the same way I run the others, it doesn't even recognize the model exists.

It is obvious I'm missing a few steps... can anyone please help?

Jaeyoung-Lim, 2022-06-30:
@oricabel Could you elaborate exactly how you were trying to run your custom model and how it is failing?

oricabel, 2022-07-01:
I tried to find the locations of other, working models (rascal, malolo) and add my model in the location, now I am sure this is not enough, perhaps the program needs to "know" that there is a new model, and if we ask to run it, it should use the models files. When I try to run my model, it just says that it does not recognize it, as if I wanted to run somthong that doesn't exist.

Jaeyoung-Lim, 2022-07-01:
@oricabel Please share exactly how you "tried" and what the "not recognized" error is. Otherwise it is hard to know what is missing from what you are describing

oricabel, 2022-07-02:
I took the model's folder named "Aeolus" (which contains it Aero model, setup file, engine, etc...) and pasted it to the following paths:
~/PX4-Autopilot/Tools/jsbsim_bridge/models/
~/PX4-Autopilot/src/px4-jsbsim-bridge/models/
~/PX4-Autopilot/Tools/flightgear_bridge/models/
These paths are where I found folders of other models (rascal, etc...).

I tried to run the simulation with the custom model using:
"make px4_sitl jsbsim_aeolus"

And received:
"ninja: error: unknown target 'jsbsim_aeolus'
Makefile:235: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1"

For reference, if I just write some letters:
"make px4_sitl jsbsim_dgf"
I get the same error.

I hope the description here is sufficient.

tayyabkhalil-313, 2022-11-08:
@oricabel You also have to add the custom model name in the cmake target list in the file _PX4-Autopilot/platforms/posix/cmake/sitl_target.cmake_ in the portion _set(models_jsbsim_. After this, you have to add a airframe for the model in the airframes directory.

raghu-srm, 2025-09-10:
@Jaeyoung-Lim 
I am seeing fcs mapping for elevator, aileron and rudder in <fcs> <channel name="All"> for rascal.,
But where thrust is being mapped? 

I am trying to develop sitl with F16 aircraft (xml's from JSBSim library)., All working fine., but even at throttle max (1) in JSBSim, engine not producing any thrust., I think somewhere mapping is not good for throttle - engine.,

I can see the clear movement of control surfaces (elevator, aileron, rudder) which all been deflected.,

kaklik, 2025-09-10:
> [@kaklik](https://github.com/kaklik) Can you share insights on this?

Unfortunately, I am not able to contribute to that because I do not have experience with px4-jsbsim-bridge.