Auterion/px4-jsbsim-bridge #60: plane can't runaway take off in HITL mode

huyandenghuangbo, 2022-10-31:
I used the pr-hitl-support branch code, I show the detail:
the command as follows:
[image]
then the px4 connect the qgc via udp, I inserted those code to read the accel and gyro  in the file of sensor_imu_plugin.cpp in realtime. Like this:
[image]
I load a fly plan into px4 and run the mission mode, but it can't runaway takeoff. And the accel and gyro data flow Like this:
[image]
I don't konw where the problem is.

Jaeyoung-Lim, 2022-10-31:
@huyandenghuangbo HIL support was added in https://github.com/Auterion/px4-jsbsim-bridge/pull/28 could you try this PR?

huyandenghuangbo, 2022-10-31:
@Jaeyoung-Lim Sorry, I'll try it right away

huyandenghuangbo, 2022-10-31:
@Jaeyoung-Lim I tried, but it can't work, the code above used comes from the pr-hil-support branch, I checked the code with #28, they are same ,but it has the same result after running. PS: I changed the code that can receive the right gps data(interface->SetHILStateLevel(true)---->interface->SetHILStateLevel(false);), as follows:
[image]

Jaeyoung-Lim, 2022-10-31:
@huyandenghuangbo Did changing the code have any effect on the results?

huyandenghuangbo, 2022-10-31:
@Jaeyoung-Lim it can't takeoff as usual
[image]

huyandenghuangbo, 2022-10-31:
@Jaeyoung-Lim I read the log file, it shows the px4's ouput actuator data has no problem
[image]

ParamDeshpande, 2022-11-07:
hey, even I'm facing similar issues here, did you guys have any luck figuring this out ? Does it have something to do with the mixer and control files ? I'm guessing the outputs of PX4 are mismatched with the outputs of the simulator so maybe the throttle doesn't really go to the motor but some other servo, so the plane does not move ? 

I was looking through this file `PX4-Autopilot/ROMFS/px4fmu_common/init.d/airframes/1000_rc_fw_easystar.hil` for HIL configuration. 

In SITL the setup works well, QGC shows some custom airframe when launching SITL, so maybe a similar config in HILStar plane might start the plane and map proper control surfaces , any Ideas ?

tayyabkhalil-313, 2022-11-08:
@ParamDeshpande Did you manage to solve this issue? I am trying to run jsbsim hitl with rascal and facing similar issue. The vehicle is unable to takeoff and instead just crashes right away. The connection from the jsbsim bridge is then closed with following error:
[image]

ParamDeshpande, 2022-11-08:
@tayyabkhalil-313 No I haven't solved the issue yet, but did you arm the plane before takeoff ?

tayyabkhalil-313, 2022-11-08:
@ParamDeshpande Yes i am able to arm the vehicle.

ParamDeshpande, 2022-11-09:
@tayyabkhalil-313 check if the plane isn't getting spawned mid-air ? Try changing locations in the `scene/LSZH.xml` file. 
```
<?xml version="1.0"?>
<initialize name="VTBS">
  <latitude type="geodetic" unit="DEG"> 13.656699 </latitude>
  <longitude unit="DEG"> 100.751837 </longitude>
  <altitude unit="M"> 0.5 </altitude>
  <elevation unit="M"> 2.5 </elevation>
  <vt unit="FT/SEC">      0 </vt>
  <gamma unit="DEG">      0.0 </gamma>
  <phi unit="DEG">        0.0 </phi>
  <theta unit="DEG">      0.0 </theta>
  <psi unit="DEG">        10.0 </psi>
</initialize>
```

The plane seems to be stable for me on VTBS airport. 
[image]

However the throttle values are still 100% and the plane doesn't seem to move. 

I tried debugging the JSBSim bridge code, by printing actuator values, in the file `src/actuator_plugin.cpp` in function `SetActuatorCommands()` 
```
      std::cout << "  actuator index: " << index << " scale: " << scale << "  property: " << property << "  value: " << scale * actuator_commands[index] << "\n";
```
For SITL things are working fine the throttle values from PX4 are passed to rascal and JSBSim, ie PX4 does send **nonzero** throttle values to the simulator and the plane takeoffs.  

[image]

However for HITL setup the PX4 hardware (CubeOrange) sends **zero** actuator values when I arm the vehicle. And a **-1.2xScale** value when it is ready to fly. 
 
[image]

This likely means some changes for the configuration files need to be done on the PX4 firmware side for the HILStar plane ? 

@Jaeyoung-Lim Do you have any ideas how to do it ? Did you manage to takeoff with the PR ?

tayyabkhalil-313, 2022-11-10:
@ParamDeshpande For me, the vehicle does move but still crashes after changing the location. 
[image]
In HITL, the throttle and rudder actuators do send non zero values but aileron and elevator do not. Throttle increases from 0 to 1 after which it crashes. 
[image]

ParamDeshpande, 2022-11-14:
Can you tell which pixhawk model are you using and the firmware version. Also if you're trying to takeoff via radio or joystick ?

tayyabkhalil-313, 2022-11-14:
@ParamDeshpande 
- I have tried on both pixhawk 4 and pixhawk 5x. 
- Firmware version is 1.12.3. 
- Trying to takeoff from mavlink console using 'commander takeoff'.

tayyabkhalil-313, 2022-11-15:
@ParamDeshpande Are you able to run JSBSim SITL for Rascal successfully? The model crashes for me in SITL as well during takeoff.

ParamDeshpande, 2022-11-15:
@tayyabkhalil-313 yeah SITL is working well for me. I think I've also managed to solve the takeoff issue in HITL too( partially atleast). I changed the config file for the `rascal.xml` config. 

Created a new config file `rascal_hitl.xml` in `configs/`
```
<model name="rascal">
    <jsbsimbridge>
        <aircraft_directory>models/Rascal</aircraft_directory>
        <aircraft_model>Rascal110-JSBSim</aircraft_model>
    </jsbsimbridge>
    <mavlink_interface>
        <tcp_port>4579</tcp_port>
    </mavlink_interface>
    <sensors>
        <imu>
        </imu>
        <gps>
        </gps>
        <barometer>
        </barometer>
        <magnetometer>
        </magnetometer>
        <airspeed>
        </airspeed>
    </sensors>
    <actuators>
        <channel name="rudder">
            <index>2</index>
            <scale>-1</scale>
            <property>fcs/rudder-cmd-norm</property>
        </channel>
        <channel name="aileron">
            <index>0</index>
            <scale>-1</scale>
            <property>fcs/aileron-cmd-norm</property>
        </channel>
        <channel name="elevator">
            <index>1</index>
            <scale>-1</scale>
            <property>fcs/elevator-cmd-norm</property>
        </channel>
        <channel name="throttle">
            <index>3</index>
            <scale>1</scale>
            <property>fcs/throttle-cmd-norm</property>
        </channel>
    </actuators>
</model>
```
The MAVLINK channels are mismatched from PX4 for Roll, pitch, yaw and throttle to the `rascal.xml`. 

I changed the config file path in `jsbsim_bridge_ros.cpp`
```
  nh_private_.param<string>("config", path, std::string(JSBSIM_ROOT_DIR) + "/configs/rascal_hitl.xml");
```

With this atleast all my actuators are responsive from PX4 to Simulator and back in HITL mode. Takeoff still crashed with mission mode on.

tayyabkhalil-313, 2022-11-15:
> I changed the config file path in jsbsim_bridge_ros.cpp

Why is that required? I think this is not used in the HITL implementation of the bridge. Instead the name of the config file can be specified while running the bridge. 
`HEADLESS=1 ./jsbsim_bridge rascal_hitl -d /dev/ttyACM0 -s ~/PX4-Autopilot/Tools/jsbsim_bridge/scene/LSZH.
xml
`
> yeah SITL is working well for me.

Did you make any changes to the rascal model or were you able to run it directly? Can you please specify the JSBSim version you are using?

ParamDeshpande, 2022-11-15:
> Why is that required? I think this is not used in the HITL implementation of the bridge. Instead the name of the config file can be specified while running the bridge.
> HEADLESS=1 ./jsbsim_bridge rascal_hitl -d /dev/ttyACM0 -s ~/PX4-Autopilot/Tools/jsbsim_bridge/scene/LSZH. xml

I'm using ROS to launch the HITL setup, but yeah this also should work.  
> Did you make any changes to the rascal model or were you able to run it directly? Can you please specify the JSBSim version you are using?

No I did not make any changes for SITL for Rascal model. 
I've installed JSBSim version `JSBSim-devel_1.2.0.dev1-935.bionic.amd64.deb` from [here.](https://github.com/JSBSim-Team/jsbsim/releases/tag/Linux)

tayyabkhalil-313, 2022-11-15:
Do you have any idea why the SITL for rascal wouldn't work with JSBSim?

ParamDeshpande, 2022-12-02:
Atleast for SITL I am able to takeoff using Mission Mode. For HITL still the plane does a barrel roll and goes inside the ground. 

I've recompiled the PX4 with these changes and uploaded them on Cubeorange, it fails. 

Can anyone check if this method works for HITL
@dagar how did you get to these parameters ? Trial and error ? Any other params to be set for HITL ?

SITL Fix : 

Find the files `1033_rascal` and `1034_rascal-electric` in PX4 Firmware and uncomment the params for the comment line 
`# fix takeoff failure for JSBsim in autonomous mission mode.`

ie set 

```
param set FW_THR_SLEW_MAX 0.3
param set-default RWTO_MAX_THR 0.6
```

herewego48, 2023-07-07:
Hello, I really need your help. I tried to run your modified code, but an error message will be reported as follows:

Opened serial device/dev/ttyACM0

In file/home/lh/Code/autopilot HITL/Tools/jsbsim_ Bridge/models/Rapid/Rapid 110 JSBSim.xml: line 104

Engine location ignored, only thrust location is used

In file/home/lh/Code/autopilot HITL/Tools/jsbsim_ Bridge/models/Rapid/Rapid 110 JSBSim.xml: line 109

Engine orientation ignored, only thrust orientation is used

Send: Connection reused

Send: Connection reused

Send: Connection reused

Send: Connection reused

If you take the time out of your busy schedule to help me take a look, I would greatly appreciate it.