Auterion/px4-jsbsim-bridge #61: HITL, No GPS Lock for Vehicle over serial

ParamDeshpande, 2022-11-03:
I have followed the instructions from this [PR](https://github.com/Auterion/px4-jsbsim-bridge/pull/28) to enable HITL support for JSB-Sim bridge and am able to connect to the JSBSim-bridge via serial port. 
I am facing the following issue in HITL mode.
`No GPS Lock for Vehicle : `

[image]

 I have selected the airplane configuration as HILStar (XPlane), on the [cubepilot cubeorange ADSB module](https://ardupilot.org/copter/docs/common-thecubeorange-overview.html). I am using the PX4-Autopilot version 1.12.3.

Any suggestions ?

ParamDeshpande, 2022-11-07:
I've solved this issue, the GPS lock can be enabled by setting 
`interface->SetHILStateLevel(false);` in the file `jsbsim_bridge.cpp` from this [PR](https://github.com/Auterion/px4-jsbsim-bridge/pull/28)

closed 2022-11-07 by ParamDeshpande

Jaeyoung-Lim, 2022-11-07:
@ParamDeshpande Great! Would it be possible to make a PR on the branch?

ParamDeshpande, 2022-11-08:
@Jaeyoung-Lim  Sure, I've created the PR on the HIL branch.