Auterion/px4-jsbsim-bridge #65: JSBSim ERROR[simulator_mavlink]Number of simulated accelerometer out of range :Max:3

brownbea, 2023-10-19:
[image]
When I try to interface JSBSim with PX4 SITL, it reports an error in this picture. But it does not report error if I try more times. It seems like random appears.
I just use make px4_sitl jsbsim in the terminal. Also I can interface gazebo with PX4 SITL successfully. Anyone has the same problem?And anyone knows how to resolve this.

Lallie101, 2023-11-23:
mavlink_interface.cpp, at around line 273, add the line sensor_msg.id = 0;

mikebevz, 2025-02-07:
It resolved the issue. However, if I have 3 IMUs then the first one is working as expected, but the second and third seem to be defunct.

[vehicle_imu] 0 - Accel: 1310988, interval: 4000.0 us (SD 0.0 us), Gyro: 1310988, interval: 4000.0 us (SD 0.0 us)
vehicle_imu: accel data gap: 0 events
vehicle_imu: gyro data gap: 0 events
ACC 1310988 EN: 1, offset: [0.000 0.000 0.000], scale: [1.000 1.000 1.000], Internal
GYRO 1310988 EN: 1, offset: [0.000 0.000 0.000], Internal

[vehicle_imu] 1 - Accel: 0, interval: 0.0 us (SD -0.0 us), Gyro: 0, interval: 0.0 us (SD -0.0 us)
vehicle_imu: accel data gap: 0 events
vehicle_imu: gyro data gap: 0 events
ACC 0 EN: 1, offset: [0.000 0.000 0.000], scale: [1.000 1.000 1.000], Internal
GYRO 0 EN: 1, offset: [0.000 0.000 0.000], Internal

[vehicle_imu] 2 - Accel: 0, interval: 0.0 us (SD -0.0 us), Gyro: 0, interval: 0.0 us (SD -0.0 us)
vehicle_imu: accel data gap: 0 events
vehicle_imu: gyro data gap: 0 events
ACC 0 EN: 1, offset: [0.000 0.000 0.000], scale: [1.000 1.000 1.000], Internal
GYRO 0 EN: 1, offset: [0.000 0.000 0.000], Internal

Any suggestion on how to fix it? Is there any design document for jsbsim-bridge?